makerbase-mks/MKS-RUMBA32 #1: BL Touch

danielbors72, 2020-03-13:
how do i connect bltouch?

makerbase-mks, 2020-03-13:
1. You can connect BLTOUCH to P15, but you may need to adjust the BLTOUCH line order to ensure normal use.
[image]
2. Add servo_pin define in pins_MKS_RUMBA32.h
[image]
3. You need enable BLTOUCH(configuration.h)
[image]

danielbors72, 2020-04-02:
if you use TMC2209 UART mode, and BLTOUCH with PD15 pin, mainboard will not work!
Why?

makerbase-mks, 2020-04-07:
This may conflict with the UART pin, so SERVO_PIN need to redefine it, such as using PB8 or PB9 pin
[image]

makerbase-mks, 2020-04-07:
@danielbors72 Sorry, I did n’t notice the problem of pin conflict at the beginning

closed 2020-06-06 by makerbase-mks

phongshader, 2020-06-27:
> This may conflict with the UART pin, so SERVO_PIN need to redefine it, such as using PB8 or PB9 pin 
>
redefine the **E1 TX pin** or the **SERVO_PIN**?

edriyin1, 2020-07-02:
Hello, I tried attaching new arrived 3D Touch sensor. I defined Servo Pin PB8 and gave 5V. But It didin't work. Can you help me?

alex18881, 2020-11-26:
Hello. I see this issue is closed, but I'm experiensing an issue with wiring my 3Dtouch. On my previous Rumba it works great but when I connect it to my new mks rumba32 it refuses to trigger x min endstop. All commands staw, deploy and self test work ok and when it starts bed leveling the bed just smashes into the nozzle.