makerbase-mks/MKS-RUMBA32 #2: TMC5160, SPI connection error

ogsv, 2020-03-15:
I am having difficulties using my TMC5160 v.1.2 drivers from bigtreetech with a v.1.0 RUMBA32 from Makerbase. I've tested many different builds of Marlin, and also the MK4duo firmware - all with the same result; "Testing X connection... Error: All LOW"

Then I came across this issue posted by repetier on the @Aus3D github page: https://github.com/Aus3D/RUMBA32/issues/17

So I guess my board also might have this issue? Any suggestions on how to fix it, would be more than welcome!

repetier, 2020-03-15:
See https://github.com/Aus3D/RUMBA32/issues/12 there chrissbar tells how to fix it with a simple bridge which is also what I did and afterwards it worked. So if MKS made a 1:1 copy they probably also copied the wrong pin positions making the jumper not work as expected.

ogsv, 2020-03-15:
Thanks a lot @repetier , problem solved! 😃 I'll leave the issue open, in case MKS should want to revise their board layout.
[image]

makerbase-mks, 2020-03-18:
Hi! Thank you for your feedback. I will add instructions for connecting to the TMC2130 or TMC5160 FAQ on the MKS RUMBA32 wiki page. Thanks again, we will improve this issue in the new version 1.1 and add BLTOUCH interface and so on

makerbase-mks, 2020-03-23:
Hi! I have created a FAQ to help users solve "TMC CONNECTION ERROR".

closed 2020-03-23 by makerbase-mks

phongshader, 2020-05-03:
Looks like there are 2 different solutions to this issue, jumping  X-RST >  SPI1  or jumping pin5 X-RST > pin6, sleep on the driver shown here https://github.com/makerbase-mks/MKS-RUMBA32/wiki/About-RUMBA32-TMC-SPI-"CONNECTION-ERROR"-FAQ. Which is correct or will both solutions accomplish the same thing? If going with the makerbase solution does that preclude using non spi drivers  like this solution does?

makerbase-mks, 2020-05-04:
Hi! @phongshader 
Their effect is the same.

phongshader, 2020-05-06:
Would this work as a solution, instead of soldering the board?
[image]
[image]

ogsv, 2020-05-06:
@phongshader , theoretically, that should also work. Now that you've done it already, why don't you tell us? :)

phongshader, 2020-05-06:
I would but I don't have the board yet.

labotecno, 2020-09-09:
> 
> 
> @phongshader , theoretically, that should also work. Now that you've done it already, why don't you tell us? :)

> 
> 
> I would but I don't have the board yet.

hello it is working like this way ?

phongshader, 2020-09-11:
it did not work but I did get a work around that did not require any soldering. all the steps are here https://old.cultivate3d.com/forums/viewtopic.php?f=6&t=299&sid=ce972bfdc1f26035f8679d28e6a662c2

studiodyne, 2021-01-28:
Or , you can also , use a thin wire , place it inside the 2 dupont connectors holes , and plug the drivers
10 sec , no tools , no heat , no electrostatic fatality
No warranty break

😄