makerbase-mks/MKS-RUMBA32 #3: Rumba32 TMC2209

dennis0212, 2020-03-30:
Hello, when i select TMC2209 then compile error in Arduino!

SanityCheck.h:2071:4: error: #error "TMC2208 Software Serial is supported only on AVR, LPC1768, STM32F1 and STM32F4 platforms."

I have TMC2209!!!!!

dennis0212, 2020-03-30:
Hello, why config and jumpering for UART on RUMBA32_MKS???  Thanks

makerbase-mks, 2020-03-31:
Hi! At present, RUMBA32 does not support TMC2209 UART mode function, which is mainly Marlin firmware. But the UART mode is set as follows:
[image]

dennis0212, 2020-03-31:
Hello, I got the TMC2209 with Rumba32 running today with the dev version 2.1.0 of Marlin and the drivers are recognized by M122 and displayed as a test ok! 
It is important to have 12V on the board, only USB outputs TMC errors
22:12:22.118 : Testing X connection... OK
22:12:22.122 : Testing Y connection... OK
22:12:22.126 : Testing Z connection... OK
22:12:22.130 : Testing E connection... OK
22:12:22.134 : Testing E1 connection... OK

Hallo, habe heute mit der dev Version 2.1.0 von Marlin die TMC2209 mit Rumba32 zum laufen bekommen und die Treiber werden bei M122 erkannt und als Test ok angezeigt! 
Wichtig ist 12V müssen am Board anliegen nur USB gibt TMC fehler aus

22:12:22.118 : Testing X connection... OK
22:12:22.122 : Testing Y connection... OK
22:12:22.126 : Testing Z connection... OK
22:12:22.130 : Testing E connection... OK
22:12:22.134 : Testing E1 connection... OK

[image]

makerbase-mks, 2020-04-01:
Can the motor run normally? Is everything normal?

dennis0212, 2020-04-01:
Yes Moror run

dennis0212, 2020-04-01:
Compile with Visual Studio Code

dennis0212, 2020-04-03:
One Problem found. When i set eeprom on this no function no eeprom fround on display.

Can help me?

makerbase-mks, 2020-04-07:
Hi! @dennis0212  STM32 series MCUs do not have EEPROM, but when the Marlin program handles this part of the function, it should be written to the flash memory using analog mode. You can see this part of the function

dennis0212, 2020-04-07:
On Version 2.0.5.3 function eprom on 2.1 no eprom. Can you this fixed ?

fjriosp, 2020-04-16:
I'm unable to get it working.

If I set any Stepper to TMC2209 (UART mode) on Marlin it stucks on boot Marlin logo.

Any help?

dennis0212, 2020-04-16:
Hello, Marlin 1.2-dev compile with Visual Studio Code with full graphik display is working on rumba32 with uart.

dennis0212, 2020-04-16:
[Marlin-dev-2.1.x.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/4489360/Marlin-dev-2.1.x.zip)

fjriosp, 2020-04-16:
I can't get it working.

Could you send me your sources?

fjriosp, 2020-04-16:
I got the full graphics working, but the steppers don't work.

Without uart the don't step. The step pin is always low.

With uart hangs on marlin logo.

dennis0212, 2020-04-17:
[Configuration.txt](https://github.com/makerbase-mks/MKS-RUMBA32/files/4490992/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-RUMBA32/files/4490993/Configuration_adv.txt)

Here my config

fjriosp, 2020-04-17:
Ok, now it works.

There is a problem in the STM32 arduino core 1.8.0 and breaks the STM32 timers so the watchdog kills the mcu in a boot loop.

Now I'm using a previous version of STM32 platform on platformio and works with serial.

Thanks!

dennis0212, 2020-04-19:
yes, you must compile with visual code on this ist no prblem

closed 2020-06-03 by makerbase-mks