makerbase-mks/MKS-RUMBA32 #6: Rumba32+Marlin2

dupaser, 2020-04-16:
Good afternoon, I have a problem with marlin2 and rumba32 mks. I can't make it work for the second day. I tried all the firmware Marlin of the second version. The problem is that the Board restarts for no reason. I tried booting a simple program to check the connection and it works.

[image]
[video-1587046434.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/4487808/video-1587046434.zip)

Bolino08f, 2020-04-18:
Hello, in configuration.h :
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1.

dupaser, 2020-04-18:
> Здравствуйте, В конфигурации.h :
> #определить SERIAL_PORT -1
> #определите SERIAL_PORT_2 1.

Bolino08f, 2020-04-20:
You have tmc2208/2209 ? In soft uart ?

dupaser, 2020-04-20:
I plan to put 2209 but when loading the Board is empty

Bolino08f, 2020-04-20:
Yes but in configuration.h is declared ?

dupaser, 2020-04-20:
I tried it in two ways yes/no

Bolino08f, 2020-04-20:
Your library of STM32 are 6.1.0 ? If yes, use ver 6.0.0

dupaser, 2020-04-20:
This is a good option. How can I view this in VS code?

dupaser, 2020-04-20:
[image]

dupaser, 2020-04-20:
I have 6.1.0, install 6.0.0

Bolino08f, 2020-04-20:
In platforms

dupaser, 2020-04-20:
Yes! Yes! This is great! Thank you very much! Works, does not restart.

closed 2020-04-20 by dupaser