makerbase-mks/MKS-RUMBA32 #9: jumpers to set for DRV8825 drivers

phongshader, 2020-06-06:
I don't see any documentation on how to set the jumpers for the drivers. If I want to set up x and y drivers to 16 micro steps and z to 32 micro steps. SLP MS3 MS2 MS1 or maybe I don't understand how this works. Documentation?

makerbase-mks, 2020-06-06:
Jumpers set like this:
[image]

phongshader, 2020-06-06:
[image: Screen Shot 2020-06-05 at 7.35.46 PM.png]

[image: Screen Shot 2020-06-05 at 7.35.37 PM.png]
If I wanted to set micro stepping 16 what jumpers would I connect? MS1=16,
MS2=32, MS3=64?

makerbase-mks, 2020-06-06:
You need acord to DRV8825 micro step set M1,M2,M3
[image]

closed 2020-06-06 by makerbase-mks

phongshader, 2020-06-06:
The pin numbers you referenced were from AUS3D board not the MKS board. the MKS pins are SLP, M2, M1, M0. [image]Would jumping SLP and M2 result in 1/16 step?

makerbase-mks, 2020-06-08:
Yes, set as your picture