makerbase-mks/MKS-RUMBA32 #12: dfu mode not recognised

phongshader, 2020-06-09:
mac:
FS mode
[image]
dfu mode
[image]

Win 10
VCP_V1.5.0_Setup_W8_x64_64bits.exe installed, it recognized the board worked once to get firmware onto board but I can no longer update the firmware.
FS mode:
[image]
dfu mode
[image]
no STM32 bootloader.
Is there a way to load firmware without dfu?

platformio:
`Uploading .pio/build/rumba32_mks/firmware.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
dfu-util: No DFU capable USB device available
*** [upload] Error 74`

makerbase-mks, 2020-06-09:
Hi! Have you viewed MKS RUMBA32 wiki? The wiki Has Tutorial about how to use MKS RUMBA32 and so on

phongshader, 2020-06-10:
I pulled out an ooold win7 netbook and it recognized the board in dfu mode. I was able to load firmware...it took several hours to compile. After that I was able to connect using a newer laptop. I don't know what happened but it doesn't seem to be a board issue

phongshader, 2020-12-01:
After working for 5 months I'm back to the same problem. I've tried several different laptops, Mac and Windows both unable to connect in DFU mode. I tried the above procedure with no success. I think the boot button is no longer functional or at least it seems to have no affect on the board when it is pressed.

mks-viva, 2020-12-02:
Abou DFU mode : https://github.com/makerbase-mks/MKS-RUMBA32/wiki/How-to-upload-marlin-firmware

Ratisse, 2020-12-02:
Power on the board and check the MCU temperature with your finger, if it is very hot, probably it has a problem. Mine had this issue (I could sometimes enter DFU mode but not always and had other issues after loading the firmware). I had to buy another board to compare, the new board MCU is cold and it works properly after loading the firmware (entering DFU mode works everytime).

phongshader, 2020-12-02:
I have tried to connect in DFU mode when the board is cold...let it sit overnight, so overheating is not an issue. I've gone through the link posted above and the boot button does nothing when pressed. When pressed in combination with the reset button the board just disconnects and never shows up in the device manager, the boot loader driver does not fail, it never tries to load because there is no device present.

mks-viva, 2020-12-03:
@Ratisse 
So, it looks more like a CPU failure?

mks-viva, 2020-12-03:
@phongshader 
Before this, your motherboard was working properly?

phongshader, 2020-12-03:
The board will still run the printer fine, I just can't update the firmware anymore. What happened was I made some changes to the firmware loaded it in DFU mode, ran some test prints, the changes I made did not fix the issue I was having, made some changes to the firmware, recompiled, and when I tried to load the new firmware I got the error "dfu-util: No DFU capable USB device available" tried a dozen different times on different usb ports on a macbook pro, and 2 different win 10 laptops. The board shows up fine in win10 and macOS but as soon as I try to go into DFU mode it disappears. Yes it was working properly.

mks-viva, 2020-12-04:
@phongshader 
That is, the motherboard is still normal, but the motherboard cannot enter DFU mode? Can i understand this way.

phongshader, 2020-12-05:
Yes the motherboard still functions but is not recognized or doesn't enter DFU mode when trying to upload firmware.

mks-viva, 2020-12-05:
So, MKS RUMBA32 wiki about 'how to uopload firmware' can it be helpful for you?

phongshader, 2020-12-05:
the problem is that the device is not seen by either macOS or windows so "We need to show STM32 BOOTLOADER installation failed" will never happen because the laptop sees no STM32 BOOTLOADER. Zadig does not see it either.

DrumClock, 2021-02-12:
Hi @mks-viva , @makerbase-mks 
first I loaded FW Marlin using STM32 BOOTLOADER and DFU mode without any problems. 
Now I wanted to upload a new FW but nothing is happening.
WIN10 Device Manager does not see the STM32 BOOTLOADER.
If I connect the board it is not a problem to communicate and send commands via USB and PRONTERFACE.

How do I load FW on a board when I don't see STM32 BOOTLOADER in the manager?

DrumClock, 2021-02-13:
Hi,
I tried another computer and it's the same.
Everything indicates that after holding the BOOT button and plugging in the USB cable the board does not invoke the DFU mod.

If I reset the board with the RESET button, it connects to the computer as COM5 
and the pronterface communicates with it without any problems.

This situation when it is not possible to run the DFU mod occurred after about 2 to 3 flash of FW Marlin.

DrumClock, 2021-02-13:
Heureka!
**the inability to start the DFU mode is caused by the connected EXP2 connector with LDC12864 display.
If the LCD 12864 disconnects the DFU mod is activated.**

PADYT0007, 2022-10-20:
HELLO I AM HAVING A PROBLEM WITH MY RUMBA 32 ... I RECEIVED AN ERROR .. ALL THE INSTRUCTIONS ARE FOLLOWED
I USE WINDOWS 11
[image]
[image]
[image]
PLS HELP!!!!!!!!!!!

DrumClock, 2022-10-21:
Hi, @PADYT0007  I don't use Merlin anymore, I switched to the better FW Klipper.
So I can't do anything for them.

DrumClock, 2022-10-22:
> > Hi, @PADYT0007 I don't use Merlin anymore, I switched to the better FW Klipper.
> > So I can't for you do nothing.
> > Can I use this FW Klipper for the rumba 32 v 1.0

Klipper is installed on Raspberry Pi, the rumba board basically serves only as a terminal for controlling the printer. read the documentation at https://www.klipper3d.org