makerbase-mks/MKS-RUMBA32 #13: G29, BLtouch

phongshader, 2020-06-11:
I've gone through the Marlin UBL setup several times, had it working on the previous MKS Rumba board. I can run G29A, G29 T, but when I run G29 P1 I get this
`1Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
[...]
Send: G29 P1
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.
Recv: 
Recv: Probing mesh point 1/100.
Recv: 
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing`
the probe deploys but nothing happens, no bed mapping, the pinter just stands idle. Is there something unique to this board that would make the Marlin documentation for UBL invalid?

phongshader, 2020-06-17:
No ideas? One thing that may be part of the problem is I'm using a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and it will lock up the printer if I use the encoder. Is this a bug or is the controller not compatible with the rumba32 board? I also have a mks tft35 that does work. Is there a way to compile the firmware without REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and just use the tft? When I try to compile without the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER I get a sanity check error and the compile fails.

makerbase-mks, 2020-06-18:
It should be compatible with LCD12864

phongshader, 2020-06-23:
has anyone gotten a bltouch and tmc5160 to work together on this board? when I issue a g29 t command the bltouch makes a squeaking noise and then goes silent. If I issue a g29 p1 command makes a squeaking noise and the printer needs to be reset. Could there be some connection to this https://github.com/makerbase-mks/MKS-RUMBA32/issues/1 where the servo pin is competing with some other resource?

phongshader, 2020-06-23:
I made the changes outlined in Aus3D/RUMBA32#8 but it made no difference 
g29 P1 still locked the printer. Any ideas?

phongshader, 2020-06-26:
I'm trying to compile one of the examples from MKS-RUMBA32-master to see if I can get this board to function with one of the included examples and I'm getting this error 
/Users/justin/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file stm32f446xe.ld: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/mks_rumba32/firmware.elf] Error 1
When I compile from Marlin-bugfix-2.0.x or Marlin-bugfix-2.0.x I do not get this error. 
What is this error?

phongshader, 2020-06-26:
After some more searching I found this https://github.com/MarlinFirmware/Marlin/issues/16088 which seems to possibly address the problem I'm having.

phongshader, 2020-07-22:
It looks as if this has been resolved in the latest Marlin 2.0 bugfix

closed 2020-07-22 by phongshader