makerbase-mks/MKS-RUMBA32 #14: EEPROM Simulation

edriyin1, 2020-06-20:
I can't use EEPROM simulation. How can I use it? In config file I defined EEPROM but M500 didn't work.

GrootOpperhoofd, 2020-07-01:
I'm having similar issues with solving this and I've tried everything with the stable version of marlin2.0.x but nothing worked. Someone pointed out that the bugfix version has some extra code included in the pins_RUMBA32_MKS.h file for the FLASH_EEPROM_EMULATION. Since there is no actual EEPROM silicon on the board it needs to be emulated in flash memory on the STM32F4. The AUS3D board of which the MKS version is derived has similar code so I'm going to try and get it to work with that.

edriyin1, 2020-07-01:
Hey @GrootOpperhoofd. I tried bugfix version of marlin today and It works. You can use emulation perfectly. But don't forget to deactivate Job Counter. It won't work with flash emulation.

GrootOpperhoofd, 2020-07-01:
Hey @edriyin1 I'm glad you got your board working. I'm still struggling to get it to work. Can you send me the configuration files that you have. I'm not sure if they will work for me since I'm trying to get a delta printer to work and those config files are a little different in some ways but knowing that you have a working setup already helps a lot for me with debugging.

edriyin1, 2020-07-01:
My setup have 2 independent Y stepper drivers and Z drivers because my cartesian printer 400mm*400mm and little bit heavy. 
[Marlin.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/4860675/Marlin.zip)

GrootOpperhoofd, 2020-07-02:
Thanks! That worked for me.

I was making this way too hard for myself. I still had some legacy stuff in my configuration from messing around with the stable marlin release. If you run the bog standard version of the latest marlin 2.0.x bugfix branch with default_envs = rumba32 and comment out //#define PRINTCOUNTER everything works like a charm both with the carthesian and with the delta configuration.

Did you already put the board in your printer? I was wondering if the TMC2209 needs to be inverted or not. I have some TMC2130's now that are inverted. Can't find any information on the 2209's

edriyin1, 2020-07-02:
I'm already using in my printer. I didn't inverted in Marlin. If I need , I am changing cables inverted.

closed 2020-07-22 by edriyin1