makerbase-mks/MKS-RUMBA32 #18: Sensorless probing with TMC2209 not working on x-motor in delta printer

GrootOpperhoofd, 2020-07-11:
Hello

I've been trying to get the MKS rumba32 to work with my delta printer. Currently everything works fine but I want to get sensorless probing to work for no other reason than it looking cool and me being to lazy to put on the probe every now and then.

I've physically connected the X-MIN, Y_MIN and Z_MIN pins on the board with some jumper wires because marlin only triggers on the Z_MIN pin and not the others. So the printer is probing the bed but it still has a little too much force on the bed making the probing a bit more unreliable than I'd like it to be. After some fiddling around with the sensitivity of stallguard I've found out that the Z and Y drivers both trip when probing but the sensitivity of the X driver is all the way up to 255 and I still don't get any false triggers. Both Y and Z are at 95 right now but this value heavily depends on the probing speed, faster probing results in values that are tuned to a higher setting.

Is there a reason the diag pin on the driver does not trigger? I've tried swapping drivers with the same result and I even have a 2nd rumba32 that shows the exact same behaviour. Since the circuitry for the 3axes is identical and two of them are working but not the 3rd, I have no other option than to concluded that the problem is in software.

Does anyone have an idea how to even start debugging this? I'd really like to get this probing to work reliably and without one of the axes not responding at all this doesn't seem to be possible.

I'm running the latest version of marlin bugfix 2.0.x with the following configuration.
[Config.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/4906607/Config.zip)

pmma, 2020-07-26:
I'm setting up a corexy printer with 6 mks tmc2099 v2 to drive X, Y, Z, E0, E1 and Z2 steppers.  All are working in UART-mode, but the sensorless probing only works on the XDIAG and YDIAG pins.  The ZDIAG and E2DIAG (Z2) did not trigger.  I tried many different setting but nothing work.  After debug the DIAG ping assignment,  I added the below definitions to the pins_RUMBA32_MKS.h and all X, Y Z and Z2 probing now works.   I'm running the latest bugfix2.0.x build.

// Sensorless DIAG pins
  #define X_MIN_PIN                         PB12
  #define E0_DIAG_PIN                       PB13
  #define Y_DIAG_PIN                        PB15
  #define E1_DIAG_PIN                       PD8
  #define Z_DIAG_PIN                        PD9
  #define E2_DIAG_PIN                       PD10

GrootOpperhoofd, 2020-07-31:
I've added the extra pin definitions to the pins_RUMBA_MKS.h but without success. Behaviour is still exactly the same. Y and Z trigger fine but X does never trigger even when at the highest sensitivity. I am running all the drivers in SPI mode but it could also be an issue for delta printers only.

pmma, 2020-08-03:
Does TMC2209 support SPI mode?  I set it to UART mode and anything work as expected.

el-quique, 2020-09-01:
Hi change pins assignament on  pins_RUMBA_common.h for a delta:
//
// Limit Switches
//
//#define X_MIN_PIN                           PB12
//#define X_MAX_PIN                           PB13
//#define Y_MIN_PIN                           PB15
//#define Y_MAX_PIN                           PD8
//#define Z_MIN_PIN                           PD9
//#define Z_MAX_PIN                           PD10

#define X_MIN_PIN                           PB13
#define X_MAX_PIN                           PB12
#define Y_MIN_PIN                           PD8
#define Y_MAX_PIN                           PB15
#define Z_MIN_PIN                           PD10
#define Z_MAX_PIN                           PD9
Work sensorless, but some rason Error on E2