makerbase-mks/MKS-RUMBA32 #19: 3D Touch With 6 TMC2209

edriyin1, 2020-07-15:
I have 6 TMC2209 in my Rumba32 board. I defined PB8 for Servo pin in pins file but It didn't work. Can you help me about that. I tried PB9 too but no luck.

edriyin1, 2020-07-19:
@chrissbarr do you have any ideas?

el-quique, 2020-08-29:
Hi. Use PD12 for signal and PB8 or PB9 for servo pin:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN PD12
The all limit pins in use for all diag_pin.
How solve TMC error on E2 (third extruder)

el-quique, 2020-09-07:
> @chrissbarr do you have any ideas?

Hi use:
in Configuration.h
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN PA9
in Configuration_adv.h
#if ENABLED(BLTOUCH)
#define BLTOUCH_DELAY 500
#define SERVO0_PIN PB8

edriyin1, 2020-09-15:
I tried that but it's not working neither. At first red led blinks but after initialization it goes blank.

DrumClock, 2021-02-14:
Hi @edriyin1 
I removed the resistors R90-R95 redefined the TX / RX pin and I can use the EXP3 connector for BL-touch 

Info here : https://github.com/makerbase-mks/MKS-RUMBA32/issues/34  
point 2 - UART mode for drivers works well and pins on conector EXP3 can be used (eg, for BL-touch, servo, etc.)

The pins PB8 and PB9 are functional for I2C bus (SDA and SCL) 
I'm not sure if they can be defined as I / O pins