makerbase-mks/MKS-RUMBA32 #21: Cannot complile firmware - stm32f446xe.ld missing

Collie147, 2020-09-01:
Trying to compile the firmware for the board.  

"Linking .pio\build\mks_rumba32\firmware.elf
%userprofile%/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f446xe.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_rumba32\firmware.elf] Error 1"

Tried with the version from issue [#3](https://github.com/makerbase-mks/MKS-RUMBA32/issues/3) which compiles ok but I can't get any serial communication on either Serial Port after upload (LED 1 also doesnt flash while it does when I use the default Firmware.bin).  Any suggestions to either are welcome, I've been banging my head off the wall on this for the last 3 days.

Thanks

el-quique, 2020-09-06:
Hi. I use Marlin 2.0.6.1 and work with 3 TMC2209. But not with 6 TMC2209

makerbase-mks, 2020-09-10:
Hi! compile firmware : https://github.com/makerbase-mks/MKS-RUMBA32/wiki/How-to-upload-marlin-firmware

Collie147, 2020-09-10:
> Hi! compile firmware : https://github.com/makerbase-mks/MKS-RUMBA32/wiki/How-to-upload-marlin-firmware

Hi, I tried that but it didnt work.  After some messing about I think its working now, I had to downgrade the ST STM32 Board Platform to version 6.0.0, every other higher version either the board didnt work or it didnt compile.  I think that should be added to the How-To document to avoid confusion for others.

Thanks

closed 2020-09-10 by Collie147