makerbase-mks/MKS-RUMBA32 #23: Several issues with brand new MKS Rumba32 board

Ratisse, 2020-09-09:
I bought this card to upgrade my CoreXY printer and replace my MKS Gen 1.4 which was working fine with the MK4duo firmware and Nextion display, But it is not so easy to make it work:

- DFU mode not recognized, solved by holding boot button while powering up the board (without reset button)

- X stepper driver is activated as soon as the board is powered up, other steppers are OK. Swapping the stepper drivers makes no difference,. No way to desactivate it but it works normally when moving on XY axis,. My stepper drivers are TMC 2208, I've tried both standalone  and UART modes, same thing!. I've tried to reflow the solders with no results

- SD card is not working. The wiring is: EXP2===> MISO=PA6, SCK=PA5, CS=PA2, MOSI=PA7  EXP1===> GND, 5V. I didn't find any other wiring possibility, maybe I missed something!! 

- Hotend fluctuations, 10° around the expected temperature, after PID autotune. Was working fine with my old card with a 1 year old firmware, had same problem with latest MK4duo firmware, so I don't know if it's a firmware or hardware issue. Hotbed is working fine (bangbang mode).

  So Im wandering if the board is faulty with so many issues on a new board or if I am faulty!

mks-viva, 2020-10-15:
- DFU mode, you can refer to wiki: https://github.com/makerbase-mks/MKS-RUMBA32/wiki/How-to-upload-marlin-firmware
- You need check your X axis EN pin IO is high or low when you power on
- SD card function, you need enable it
- Does Hotend have a cooling fan that affects temperature measurement?

Ratisse, 2020-10-16:
- X axis EN pin IO is shorted to ground with nothing connected and not powered.
- Powering  the card with nothing else connected, it takes 500mA and MCU is very hot, about 65°C (checked with thermal camera)
- Ordered and received a new card, everything listed above is OK. Powering the card with nothing else connected, the new card takes 100mA and MCU is cold
- New card attached to the printer, everything works fine

mks-viva, 2020-10-20:
@Ratisse Is everything working properly?

Ratisse, 2020-10-20:
Yes, with the new one everything is OK

mks-viva, 2020-11-28:
Ok! I will close this issues.

closed 2020-11-28 by mks-viva