makerbase-mks/MKS-RUMBA32 #24: Can't find board via USB using a raspberry

Bernouillle, 2020-10-05:
Hello there,

i just purchased one of your MKS Rumba32 boards and i have problems to communicate with this board via USB. I am using the firmware Klipper, which uses a Raspberry Pi as a host computer. This host computer should communicate with the Rumba32 board via USB, but the board is not shown as connected to the Raspberry. Can you help me with this?
Best regards!

mks-viva, 2020-10-15:
If use marlin firmware and it's will work?

Bernouillle, 2020-10-26:
I can't use any other firmware than Klipper, that i need for my work. If the USB connection doesn't work, can you explain to me how i set up a uart connection between the MKS Rumba32 and a Raspberry Pi 3b+ ?