makerbase-mks/MKS-RUMBA32 #25: where i can add wifi and nextion screen ?

labotecno, 2020-11-06:
hello , i need to know where i can add this element ? wifi and touch screen ?

mks-viva, 2020-11-24:
Hi! You connect MKS TFT touch screen display by AUX-1 interface and MKS RUMBA32 have not wifi interface.

labotecno, 2020-11-24:
why mks rumba 32 don t have wifi ? how i can add it ?

mks-viva, 2020-11-24:
I want to know, which WIFI module do you want to add? Usually WIFI is the main method of serial communication, such as ESP8266(ESP3D WIFI firmware), so you can connect it to STM32 Usart1 or other pin
[image]

labotecno, 2020-11-24:
ok perfect ! and marlin stuff ? serial ? i need to change pin on board pin ?

mks-viva, 2020-11-24:
You just need to pay attention to your pin and motherboard one-to-one correspondence. For example,like this:
WIFI module's TX connect to motherboard's RX(PA10 STM32 USART1 RX)
WIFI module's RX connect to motherboard's TX(PA9 STM32 USART1 TX)
WIFI module's 5V connect to motherboard's 5V
WIFI module's GND connect to motherboard's GND
More information you can refer to MKS SGEN_L V2 Wiki how to use MKS TFT WIFI for marlin2.0 and fimrware settings
Link： https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_TFT_WIFI

mks-viva, 2020-12-01:
Has this issues be solved? If there is no response, then I will close it temporirly.

closed 2020-12-01 by mks-viva

labotecno, 2020-12-02:
yes its ok thank you