makerbase-mks/MKS-RUMBA32 #26: heater errors

phongshader, 2020-11-09:
this problem just started. I have a large format printer that I switched over to a .60 nozzle to print some 420x420x100 prints @ 230°, prior to that I had no problems printing with a .40 nozzle @ 210°. After switching to the larger nozzle and higher temp I started getting heater errors. I tested/switched out thermistors, heater cartridges, checked all the wiring, ran PID tunes. This appeared fix it until I ran a test print to verify...it failed again. The HE heat to 230° then cool to 219° heat to around 222°, then with the heater led still lit the temp would drop until the firmware halted the printer. I finally got out the voltmeter tested the output on the HE0 output. I get a full 12 volts on the initial heat up and it would pulse to regulate the temp 1 time then fail...what is confusing is that the output would drop from 230 to 219 and go back to 12v up to 222 then the voltage would drop off until the firmware halted the printer but the heater led remained lit...so the board still thinks it's heating when it's not? I haven't tried switching the heater over to HE1, that's next. Could this be a mosfet failure? If so do you have recommendations on a replacement mosfet that I could find in the US?

phongshader, 2020-11-10:
switching to HE1 seemed to fix the problem...so it looks like the mosfet on the board is bad...unless you all have a better idea. Any suppliers in the US or alternative mosfet for the HY1403?

mks-viva, 2020-11-24:
Hi! You can change it to other interface, HY1403 is a N channel MOS 42A/30V, so, if you change it, you can by similar parameters MOS

phongshader, 2020-11-24:
I did change to HE1 and that seemed to fix most of the issues...but I'm still having a heater problem, when I run an M600 and the temp drops below the target temp, expected behavior, when the new filament is loaded and the heater attempts to reach the target temp it never does, that is it get to within 7-10 degrees and never gets closer and then the printer eventually halts on a thermal runaway error. However if I reset the printer it has no problem reaching the same temp. I don't know if this is a hardware or firmware issue but I've lost several prints after 18+ hrs on a print.

mks-viva, 2020-11-25:
Ok! According to your statement, I have to doubt the heat preservation treatment of the printer nozzle. Many machines now do this special treatment to ensure that the nozzle heat is not lost and the temperature can be constant. This is the customer machine, I think the nozzle temperature protection Very necessary
[image]

phongshader, 2020-11-25:
I do have a silicone sock on the heater block. I don't believe this is a heat retention issue, this is a heater control issue. As I said in my previous post during a filament change the heater cools down by about 10-15 degrees, which is to be expected, and when the filament is loaded the heater tries to warm up but cannot achieve it's desired target and the printer halts in the middle of a print...however when I reset the printer after the halt it has no problem reaching the desired temp. This would lead me to believe that there is a control issue either in hardware or firmware because the heater is clearly able to heat to the target temp but for some reason if that temp drops below a certain temp while printing it is unable to recover. You might say that's because there is filament being pushed through the nozzle or the cooling fan is too aggressive, but this happens when the printer is warming up after a filament change when there is no part fan running or filament running through the nozzle.
[image]

mks-viva, 2020-11-26:
This phenomenon seems incomprehensible. According to normal operation, PID is usually set, the heating head has heat preservation measures, and the temperature should be in a constant state.

phongshader, 2020-11-26:
Yes it is incomprehensible and yet it is happening. I have set PID, with the cooling fan @ 100%...no problem maintaining the set temp while tuning.

mks-viva, 2020-11-26:
What is the power supply? I want to know, whether the heating bed is heated and FAN is run 100%, when we are setting the PID of the heating headend

phongshader, 2020-11-26:
here is the temp graph for a filament change
[image]

phongshader, 2020-11-26:
it looks like the PID tuning is trying to tune to 210 when the printer is commanding 220. I don't know where 210 is coming from. If I get the filament changed before the temp drops more than 5 the the printer resumes as expected after a M600 command.
[image]

mks-viva, 2020-11-27:
Ok! I got your thoughts. You want to set the temperature range of PID adjustment at plus or minus 5℃
About this function, marlin's Configuration.h file can be set:
#define PID_FUNCTIONAL_RANGE 10
Default is 10, you can set it to 5
[image]

phongshader, 2020-11-27:
I set **#define PID_FUNCTIONAL_RANGE**  to 5 as you suggested. That stopped the printer from halting but I did not fix the bigger problem. The heater stays within the thermal runaway limits so it does not halt but the heater still never reaches the target of 220. It will heat up to within 5 of the target as specified with **PID_FUNCTIONAL_RANGE 5** but then the PID kicks in and acts like 215 is the target temp not 220 so the heater never reaches 220, the closest it gets is 217, and the print never starts again.
[image]

phongshader, 2020-11-27:
configuration.h:
`#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

/**
 * Max Bed Power
 * Applies to all forms of bed control (PID, bang-bang, and bang-bang with hysteresis).
 * When set to any value below 255, enables a form of PWM to the bed that acts like a divider
 * so don't use it unless you are OK with PWM on your bed. (See the comment on enabling PIDTEMPBED)
 */
#define MAX_BED_POWER 255 // limits duty cycle to bed; 255=full current

#if ENABLED(PIDTEMPBED)
  //#define MIN_BED_POWER 0
  //#define PID_BED_DEBUG // Sends debug data to the serial port.

  // 120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  // from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 190.05
  #define DEFAULT_bedKi 37.12
  #define DEFAULT_bedKd 648.72

  // FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
#endif // PIDTEMPBED

#if EITHER(PIDTEMP, PIDTEMPBED)
  #define PID_DEBUG             // Sends debug data to the serial port. Use 'M303 D' to toggle activation.
  //#define PID_OPENLOOP          // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
  //#define SLOW_PWM_HEATERS      // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
  #define PID_FUNCTIONAL_RANGE 5 // If the temperature difference between the target temperature and the actual temperature
                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.
#endif`

mks-viva, 2020-11-28:
Whether to try to re-tune the PID？

phongshader, 2020-11-28:
I tuned several times with the same result

mks-viva, 2020-11-30:
At this stage, we can only use testing to try, whether it can solve the problem.
For example, set #define PID_FUNCTIONAL_RANGE 15 and test it

phongshader, 2020-12-06:
unfortunately I can't test at the moment because of this https://github.com/makerbase-mks/MKS-RUMBA32/issues/12

mks-viva, 2020-12-07:
It seems there is still a lot to be solved.