makerbase-mks/MKS-RUMBA32 #27: MKS OLED 0.96 on SSD1306 not working

alex18881, 2020-11-24:
Recently I decided to upgrade my printer mainboard from BIQU Rumba to MKS Rumba32. And now I can't manage the LCD to work. The same Marlin with just mainboard changed on RUMBA works great with my old Rumba and this LCD. With Rumba32 I can hear a beeper sound when I click the encoder, just as it suppose to be. The integrated in LCD SD card works as well: M20 command returns a list of files on SD card in the Platformio terminal. The only issue is an absolutely black screen. Also when I connect my old discount 2004 text smart controller it works like a charm.
All stepper drivers (TMC2208) are connected in step/dir mode.

* MKS Rumba32
* MKS OLED 0.96 2015.11.8 v1.0

Configuration.h:
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_RUMBA32_V1_0
#define EXTRUDERS 1
#define PIDTEMP
#define PIDTEMPBED
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define BLTOUCH
#define SERVO0_PIN PB8
#define AUTO_BED_LEVELING_BILINEAR
#define Z_SAFE_HOMING
#define LCD_LANGUAGE ru
#define DISPLAY_CHARSET_HD44780 CYRILLIC
#define LCD_INFO_SCREEN_STYLE 0
#define SDSUPPORT
#define MKS_12864OLED_SSD1306
#define NUM_SERVOS 1
#define SERVO_DELAY { 300 }
```

pins_RUMBA,h (as a pins reference)
```
#define LED_PIN                               13
#define PS_ON_PIN                             45
#define KILL_PIN                              46
...
...
#if EITHER(MKS_12864OLED, MKS_12864OLED_SSD1306)
  #define LCD_PINS_DC                         38  // Set as output on init
  #define LCD_PINS_RS                         41  // Pull low for 1s to init
  // DOGM SPI LCD Support
  #define DOGLCD_CS                           19
  #define DOGLCD_MOSI                         42
  #define DOGLCD_SCK                          18
  #define DOGLCD_A0                  LCD_PINS_DC
...
#else
  #define LCD_PINS_RS                         19
  #define LCD_PINS_ENABLE                     42
  #define LCD_PINS_D4                         18
  #define LCD_PINS_D5                         38
  #define LCD_PINS_D6                         41
#endif

#define LCD_PINS_D7                           40

//
// Beeper, SD Card, Encoder
//
#define BEEPER_PIN                            44
```

pins_MKS_RUMBA32.h
```
  #define BTN_EN1                           PB2
  #define BTN_EN2                           PB1
  #define BTN_ENC                           PE7

  #if EITHER(MKS_12864OLED, MKS_12864OLED_SSD1306)
    #define LCD_PINS_DC                     PE13  //38 // Set as output on init
    #define LCD_PINS_RS                     PE14  //41 // Pull low for 1s to init
    // DOGM SPI LCD Support
    #define DOGLCD_CS                       PE10  //19
    #define DOGLCD_MOSI                     PE9   //42
    #define DOGLCD_SCK                      PE12  //18
    #define DOGLCD_A0                       LCD_PINS_DC
  
  #else

    #define LCD_PINS_RS                     PE10
    #define LCD_PINS_ENABLE                 PE9
    #define LCD_PINS_D4                     PE12

    #if ENABLED(MKS_MINI_12864)
      #define DOGLCD_CS                       PE13
      #define DOGLCD_A0                       PE14
    #endif

    #if ENABLED(ULTIPANEL)
      #define LCD_PINS_D5                     PE13
      #define LCD_PINS_D6                     PE14
    #endif

    #define LCD_PINS_D7                     PE15 // 40
  #endif

```

mks-viva, 2020-11-24:
Hi! You need to pay attention to the pin define.

alex18881, 2020-11-24:
Yes, I totally understand this. Thank you for such quick response. According to this [RUMBA32 pin scheme](https://github.com/makerbase-mks/MKS-RUMBA32/blob/master/hardware/MKS%20RUMBA32%20V1.0_002/MKS%20RUMBA32%20V1.0_002%20PIN.pdf) and this [RUMBA+ pin scheme](https://github.com/Aus3D/RUMBA-Plus/blob/master/Resources/Images/PinoutDrawing.png) pins definition is ok. As I can see in Marlin code Rumba+ and Rumba have the same pinout.

mks-viva, 2020-11-24:
Ok! Can you provide your firmware to me and i will test it?

alex18881, 2020-11-24:
Yes. Now I switched to fork from original MarlinFirmware here https://github.com/alex18881/Marlin/tree/2.0.x-rumba32. If this matters I can make the same changes in fork from this FW.

mks-viva, 2020-11-24:
Okay, so I can also directly download the official marlin to test the MKS OLED display function?

alex18881, 2020-11-24:
I'm not sure I correctly understand the question. But this screen works fine with the original Marlin on 8bit RUMBA and RAMPS. If you have this hardware you can check how they work.

mks-viva, 2020-11-24:
Ok! Waiting for my good news.

alex18881, 2020-11-24:
The interesting thing according to schematics in this repository and in the Aus3d's RUMBA+ exp1 and exp2 have absolutely the same pinout. All pins are the same.

mks-viva, 2020-11-25:
Hi! I have test offical marlin fix bug 2.0.x and MKS OLED have not display information, but i use 2560 and MKS OLED will display information. So, i think MKS OLED run to MKS RUMBA32 have not display information is due to STM32 lib not perfect, or caused by poor compatibility. We expect offical marlin to be compatible with MKS OLED on MKS RUMBA32

alex18881, 2020-11-25:
In my research I came to the same conclusion. But there are several posts and articles in internet where people make oled displays to work on stm32 based mainboards (most of them biqu ones). That were not this MKS oled but plain 4 pins displays on ssd1306. And they work somehow. That was why I thought that I'm doing something wrong if my one doesn't work. 
Also I read there's a gcode which can help determine what pin exactly is triggered. May be there's a mistake in pin definitions?

mks-viva, 2020-11-25:
I have reported this issues to the marlin.