makerbase-mks/MKS-RUMBA32 #28: Build Failed

RepRapThailand, 2020-11-27:
Hi.

Atom 1.52.0 PlatformIO 5.0.3

Compiling .pio\build\mks_rumba32\FrameworkArduino\stm32\usb\usbd_desc.c.o
C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:46:4: error: #error "USB VID or PID not specified"
   46 |   #error "USB VID or PID not specified"
      |    ^~~~~
In file included from C:\Users\krp\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc/usbd_core.h:30,
                 from C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:21:
C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:160:10: error: 'USBD_PID' undeclared here (not in a function); did you mean 'USBD_VID'?

mks-viva, 2020-11-27:
Hi! I think this is caused by some libraries on the compilation platform. At this stage, people are more willing to use Vscode to compile firmware.

RepRapThailand, 2020-11-27:
Which version should I install, both VSC and PIO?

mks-viva, 2020-11-27:
Like this: https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/Marlin_firmware

RepRapThailand, 2020-11-27:
I try VScode 1.51.1 and PIO 2.2.1 and same error.

RepRapThailand, 2020-11-27:
Building in release mode
Compiling .pio\build\mks_rumba32\FrameworkArduino\stm32\usb\usbd_desc.c.o
Compiling .pio\build\mks_rumba32\FrameworkArduino\stm32\usb\usbd_ep_conf.c.o
C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:46:4: error: #error "USB VID or PID not specified"
   46 |   #error "USB VID or PID not specified"
      |    ^~~~~
In file included from C:\Users\krp\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc/usbd_core.h:30,
                 from C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:21:
C:\Users\krp\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\usbd_desc.c:160:10: error: 'USBD_PID' undeclared here (not in a function); did you mean 'USBD_VID'?
  160 |   LOBYTE(USBD_PID),           /* idProduct */
      |          ^~~~~~~~
C:\Users\krp\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc/usbd_def.h:275:32: note: in definition of macro 'LOBYTE'
  275 | #define LOBYTE(x)  ((uint8_t)((x) & 0x00FFU))
      |                                ^
*** [.pio\build\mks_rumba32\FrameworkArduino\stm32\usb\usbd_desc.c.o] Error 1
================================================================================================== [FAILED] Took 5.53 seconds ==================================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_rumba32    FAILED    00:00:05.533
============================================================================================= 1 failed, 0 succeeded in 00:00:05.533 =============================================================================================
The terminal process "C:\Users\krp\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'mks_rumba32'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

mks-viva, 2020-11-27:
Did you modify the firmware? Can you provide your firmware to me?

RepRapThailand, 2020-11-27:
I have not modified. Link  https://github.com/makerbase-mks/MKS-RUMBA32/tree/master/Firmware/Marlin-bugfix-2.0.x

RepRapThailand, 2020-11-27:
I try  compile on another computer and same error.

RepRapThailand, 2020-11-27:
Do I have to connect the board while build?

mks-viva, 2020-11-27:
Haha! You can download offical marlin and change it, MKS RUMBA32 path marlin firmware have not update to the latest version marlin firmware.

RepRapThailand, 2020-11-27:
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:776:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
  776 |     #error "Unknown MOTHERBOARD value set in Configuration.h"
      |      ^~~~~
CalledProcessError: Command '"C:\Users\krp\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DSTM32F4xx -DARDUINO_RUMBA32_F446VE -DARDUINO_ARCH_STM32 -DBOARD_NAME=\"RUMBA32_F446VE\" -DSTM32F446xx -DUSBCON -DUSBD_VID=0x8000 -DUSB_MANUFACTURER=\"Unknown\" -DUSB_PRODUCT=\"RUMBA32_F446VE\" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\krp\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(item, exports="env")
  File "C:\Users\krp\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\krp\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\krp\Documents\PlatformIO\Projects\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 287:
    apply_features_config()
  File "C:\Users\krp\Documents\PlatformIO\Projects\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\krp\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\krp\Documents\PlatformIO\Projects\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 263:
    load_marlin_features()
  File "C:\Users\krp\Documents\PlatformIO\Projects\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 251:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\krp\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\krp\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)

RepRapThailand, 2020-11-27:
[image]

mks-viva, 2020-11-27:
You need define BOARD_RUMBA32_MKS
[image]

RepRapThailand, 2020-11-27:
[image]

RepRapThailand, 2020-11-27:
[image]

RepRapThailand, 2020-11-27:
Back to the same failure.

mks-viva, 2020-11-27:
What screen are you using? I test a firmware and provide it to you

RepRapThailand, 2020-11-27:
MKS TFT35 V1.0 and RepRapDiscount Full Graphic Smart Controller.

RepRapThailand, 2020-11-27:
Which version do you use for VScode and Platformio?

mks-viva, 2020-11-27:
[update firmware bin file.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/5607622/update.firmware.bin.file.zip)
1. Download it
2. MKS RUMBA32 enter DFU mode(Press down BOOT ， connect to PC by USB)
[image]
3. Run Upload.bat
[image]
[image]
4. Test is ok
[RUMBA32+TFT35+LCD12864 TEST.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/5607650/RUMBA32%2BTFT35%2BLCD12864.TEST.zip)

mks-viva, 2020-11-27:
The firmware link is: https://github.com/makerbase-mks/MKS-RUMBA32/blob/master/Firmware/Marlin-2.0.7.2.zip

RepRapThailand, 2020-11-27:
Successful with your upload.bat but TFT35 not responding,  full graphic smart controller working.

RepRapThailand, 2020-11-27:
[image]

RepRapThailand, 2020-11-27:
[VID_20201127_215531.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/5608422/VID_20201127_215531.zip)

RepRapThailand, 2020-11-27:
> The firmware link is: https://github.com/makerbase-mks/MKS-RUMBA32/blob/master/Firmware/Marlin-2.0.7.2.zip

What version of VScode and Platformio are you using?

I use 1.51.1 and 2.2.1 and st stm32 9.0.

mks-viva, 2020-11-28:
> > The firmware link is: https://github.com/makerbase-mks/MKS-RUMBA32/blob/master/Firmware/Marlin-2.0.7.2.zip
> 
> What version of VScode and Platformio are you using?
> 
> I use 1.51.1 and 2.2.1 and st stm32 9.0.

Haha! I use MKS-Marlin2.0-Compile.exe : https://github.com/makerbase-mks/MKS-Marlin2.0-Compile-Tool

mks-viva, 2020-11-28:
> Successful with your upload.bat but TFT35 not responding, full graphic smart controller working.

You need set MKS TFT35 baudrate to 250000 by config file

RepRapThailand, 2020-11-28:
> > Successful with your upload.bat but TFT35 not responding, full graphic smart controller working.
> 
> You need set MKS TFT35 baudrate to 250000 by config file

yes, baudrate 250000

[image]

RepRapThailand, 2020-11-28:
ok, success!
ขอบคุณมาก

closed 2020-11-28 by RepRapThailand