makerbase-mks/MKS-RUMBA32 #29: All Endstops not working

gsart84, 2020-12-03:
Hello

sorry for my bad english

i have a MKS Rumba32 V1.0 Board and i loadet Marlin from MKS github.
All Endstop does not work.

Previously there was a old Rumba Board in my Printer ans all endstops working fine.

can anyone help? 

i attached the files

[Marlin.zip](https://github.com/makerbase-mks/MKS-RUMBA32/files/5633904/Marlin.zip)

mks-viva, 2020-12-04:
Hi! Ok i will check it tomorrow.

gsart84, 2020-12-04:
Hello

got it.

Bending the diag pin on stepper driver solve the problem!

mks-viva, 2020-12-05:
Yes, TMC2209 will output a singal by diag pin, MKS RUMBA32 connected it to endstop signal pin by 0603-0Ω
[image]
[image]

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva