makerbase-mks/MKS-RUMBA32 #31: [Q] pinout and button ?

DrumClock, 2021-01-15:
Hi, I have a few questions:

1) What is the "USER" button for?
2) can the points on the PCB marked TP3, TP4, and TP5 be used as additional I / O pins?

Thank you.

mks-viva, 2021-01-16:
Hi!
1. User button you can define it as kill_pin
2. TP3: PB10，TP4: PA15，TP5: PD11，you can define it to other function

DrumClock, 2021-01-18:
HI, thank you very much.
Is it possible to define TP3-TP5 pins as PWM for servo control use?

mks-viva, 2021-01-19:
Hi!
PB10/PA15 is PWM pin

DrumClock, 2021-01-27:
Thanks @mks-viva , I have one more question.

If I use the TMC 2209 UART-Mode driver, will I be able to use the I / O pins on the EXP 3 connector
 for other functions (eg 2x servo, Neopixel LED, etc.)?

mks-viva, 2021-01-29:
No, because it has been used in TMC2209 uart

DrumClock, 2021-01-29:
Hi, I also noticed it while studying the schematic.

If I compare it, for example, with the board ROBIN_PRO
 which uses` SERIAL_TX_PIN = SERIAL_RX_PIN .... PD5`

Query:

What if I remove resistors R90 to R95 and define them
same `SERIAL_RX_PIN = SERIAL_TX_PIN .... PC14`

Then I could use EXP3 without limiting the UART mode.

Thank you

mks-viva, 2021-01-30:
Yes, it can be done. Currently STM32 supports 1 IO for UART mode setting

DrumClock, 2021-01-30:
Hi @mks-viva 
OK, I'm waiting for the delivery of the MKS Rumba32 board. 
I'll run a test and give info on how it works.

mks-viva, 2021-02-01:
But you need test it. So, i will close this issues temporarily.

closed 2021-02-01 by mks-viva

DrumClock, 2021-02-12:
@mks-viva 
Please help,
I can't access the record to FW Marlin.
The device is installed in the manager, but when I connect the board,
Win10 does not load the STM32 BOOTLOADER driver. 

[image]

method:
1. connect the Rumba board via USB to the computer
2. press the RESET button together with BOOT

DrumClock, 2021-02-12:
so the STM32 BOOTLOADER no longer appears at all in the device manager ....