makerbase-mks/MKS-RUMBA32 #33: BUG - MKS Rumba32 and Bigtreetech TFT 35 E3 V1.0

DrumClock, 2021-02-13:
info: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1624

DrumClock, 2021-02-14:
It is said that this is not a problem of Bigtreetech TFT panel but incompatibility of MKS Rumba32
 .... see communication on the link above ...

mks-viva, 2021-02-14:
Hi, you connect it by Aux-1 interface

DrumClock, 2021-02-14:
The AUX-1 interface only works from TFT mode and some functions are inaccessible (eg IDEX mode).
Therefore I also have connected EXP1 and EXP2 interfaces for Mod LCD 12864.

There is a problem that the MKS uses pin PB2 (BOOT1) as pin BTN and it is a problem to switch the board to DFU mode because this pin is permanently in H (3.3V) from the TFT panel.

The solution is to remap both HW and SW pin BTN to another.

closed 2021-03-29 by DrumClock