makerbase-mks/MKS-RUMBA32 #34: INFO -  modifications Rumba32 board

DrumClock, 2021-02-13:
Hi, below are possible modifications :

1) **use jumpers for DIAG** settings instead of R82-R87
   possibility to connect jumper to Endstop or wire as needed e.g.

     XDIAG = X- (jumper)
     YDIAG = Y- (jumper)
     ZDIAG = Z + (wire for Z-max position)
     E2DIAG = Y + (wire, for IDEX mode)

2) **remove R90-R95** for UART-mod use:

 ```
#define X_SERIAL_TX_PIN PC14
 #define X_SERIAL_RX_PIN PC14

  #define Y_SERIAL_TX_PIN PE4
  #define Y_SERIAL_RX_PIN PE4
  
  #define Z_SERIAL_TX_PIN PE0
  #define Z_SERIAL_RX_PIN PE0

  #define E0_SERIAL_TX_PIN PC13
  #define E0_SERIAL_RX_PIN PC13
 
 #define E1_SERIAL_TX_PIN PD5
  #define E1_SERIAL_RX_PIN PD5
 
  #define E2_SERIAL_TX_PIN PD1
  #define E2_SERIAL_RX_PIN PD1
```

3) **create an EXP 4** connector
     pin 1,3,5,7,9 = BL Touch
     pin 4,6,8 = Neopixel
     pin 2 = external BOOT switch
     pin 10 = external USER switch
  
 ```
 GND -    1     2 - BOOT0 (BOOT Button)
 + 5V -   3     4 - + 5V
 PA15 -   5     6 - PB10
 GND -    7     8 - GND
 PD11 -   9    10 - PC10 (USER Button)

```

###    **I will test everything and give information**

DrumClock, 2021-02-14:
### Hi, first tests:

**point 1)** - StallGuard cannot be set for Z axis in Marlin FW - others work.

**point 2)** - UART mode for drivers works well and pins on conector EXP3 can be used (eg, for BL-touch, servo, etc.)

mks-viva, 2021-02-14:
Thank you, we will consider it.

DrumClock, 2021-02-14:
OK and change the pin for BTN to other than PB2 (BOOT1) !!

DrumClock, 2021-02-14:
The Rumba board (mega2560 or STM32) is, in my opinion, the most equipped for various variants of 3D printers.

So far I have not found any that would allow 6 motors, 5 temperature sensors, and 5 HE / FAN switches .....

I would just make a few HW changes and it will be the best board ....

DrumClock, 2021-02-14:
### Here is my suggestion. 

[image]

FinnsMax, 2021-02-18:
Hey, what are the R90-R95 resistors on the board for? I want to use tmc2209 uart, but I do not plan such a deep redesign of the board. Can resistors immediately delete and reassign the pins?

DrumClock, 2021-02-18:
Hey @FinnsMax 
If you do not want to use the EXP3 connector with the TMC2209-UART mode as well, you do not need to do anything.

information: https://github.com/makerbase-mks/MKS-RUMBA32/issues/31#issuecomment-768449817

FinnsMax, 2021-02-18:
Ок

Cobactan57, 2021-02-20:
Hi, in the current setting could you point a pin I could use as data pin for neopixel ? Thanks !

DrumClock, 2021-02-20:
> Hi, in the current setting could you point a pin I could use as data pin for neopixel ? Thanks !

Hi, the description is how I created a new EXP4 connector for BL-touhc and NeoPixel.
It's tried and tested.

Cobactan57, 2021-02-20:
Thanks, but sorry it's not clear to me: can I use PB10 as neopixel pin without modifying to get an Exp4 ? If so where is PB10 on the board ? Is it TP3 ?  How about using PD14 on Exp3 ? Many thanks !

DrumClock, 2021-02-20:
> Thanks, but sorry it's not clear to me: can I use PB10 as neopixel pin without modifying to get an Exp4 ? If so where is PB10 on the board ? Is it TP3 ? How about using PD14 on Exp3 ? Many thanks !

1)  TP3: PB10，TP4: PA15，TP5: PD11，you can define it to other function

2)  PD12,PD13,PD14, PD13, on EPX3 you can also use but only if you do not have TMC2209 UART mode.

Cobactan57, 2021-02-20:
Will try, Thanks a lot !

DrumClock, 2021-02-20:
I have tried it ..... it works :)

FinnsMax, 2021-03-09:
[image]
[image]
[image]

FinnsMax, 2021-03-09:
Servo0 --> TP4 --> NC AUX1
100 K -->B2 EXP2 (GND) --> https://github.com/Aus3D/RUMBA32/blob/master/Resources/Images/RUMBA32_v1_1B_Schematic.png

FinnsMax, 2021-03-09:
With TFT has not yet tried. Works with LCD RepRap

FinnsMax, 2021-03-09:
I have Hi only of BTT TFT35 V3.0, will try soon

DrumClock, 2021-03-14:
> Thank you, we will consider it.

Hi,
will you be releasing a new version of Rumba32 soon?
Or should I buy version 1.0 and remake it again?

T0bbi3, 2021-04-19:
Ive been all in for buying rumba32 as i got other mks products. But for now I'll wait some time for a new revision of the board then buy other brand if nothing is released. Doesn't seem to be any huge faults but I'm no good in fine electronics/soldering. People will pay more for a fixed board. Please give some release info on this🙈

DrumClock, 2021-05-17:
Hi @mks-viva 
BIGTREETECH has released a new OCTOPUS-V1.0 board.
Are you also planning to use a board with these parameters?
If so, BTT lacks a connector for connecting two other servos (except BLTouch)

closed 2022-01-30 by DrumClock